haiibo/OpenWrt #1: 今天最新的armv8，贝壳云的固件，argon主题消失

Benjamin23333, 2021-12-19:
我的贝壳云今天更新了最新的plus版本固件
[image]
但是argon主题不见了，我以为是您的配置文件更改了，我就看了下并没改动主题的部分，但是软件包里确实没有argon主题了
[image]
可以问下是怎么回事吗？

haiibo, 2021-12-20:
应该是上游有更改，我看一下

Benjamin23333, 2021-12-20:
> 应该是上游有更改，我看一下

我觉得有必要解释下，以防您误会，我在argon主题的项目下面提的issue是我在你的固件上安装了argon的ipk，之后出现的问题

haiibo, 2021-12-20:
[image]
名字改变了

haiibo, 2021-12-20:
已修改并重新编译完成，重新更新即可

Benjamin23333, 2021-12-20:
> 已修改并重新编译完成，重新更新即可

我试一下

Benjamin23333, 2021-12-20:
> 已修改并重新编译完成，重新更新即可

好了，而且bing壁纸也能用了

closed 2022-01-28 by Benjamin23333