haiibo/OpenWrt #8: 怎么把这么好用的镜像打包成Docker镜像发到DOCKER网上站去？这个镜像太牛了

lanyucomputer, 2022-04-19:
怎么把这么好用的镜像打包成Docker镜像发到DOCKER网上站去?然后用DOCKER PULL命令拉取，这样就可以造福更多的人了.这个镜像太牛了，还有其它芯片的可以加进来吗？

lanyucomputer, 2022-04-22:
docker pull lanyucomputer/zyxq:latest 这个是我自己用的你镜像打包的docker 镜像，就是有点大，可以用试试看

star2k, 2022-05-17:
> docker pull lanyucomputer/zyxq:latest 这个是我自己用的你镜像打包的docker 镜像，就是有点大，可以用试试看

有docker部署指令么？

lanyucomputer, 2022-05-26:
> 

有命令的，有一点麻烦，要ubuntu或类似环境，https://www.vediotalk.com/archives/34583

lanyucomputer, 2022-05-26:
> > docker pull lanyucomputer/zyxq:latest 这个是我自己用的你镜像打包的docker 镜像，就是有点大，可以用试试看
> 
> 有docker部署指令么？

有命令的，有一点麻烦，要ubuntu或类似环境，https://www.vediotalk.com/archives/34583

dizicheng, 2022-05-26:
img怎么打包成docker啊，教程都是tar.gz的

closed 2022-07-17 by haiibo

Java2Boy, 2023-06-22:
> > > docker pull lanyucomputer/zyxq:latest 这个是我自己用的你镜像打包的docker 镜像，就是有点大，可以用试试看
> > 
> > 
> > 有docker部署指令么？
> 
> 有命令的，有一点麻烦，要ubuntu或类似环境，https://www.vediotalk.com/archives/34583

有点大啊 老铁，有MINI版 Docker镜像么？求更新