haiibo/OpenWrt #22: Issue with ethernet and wireless controller

umarizulkifli, 2022-08-15:
```
Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL810xE PCI Express Fast Ethernet controller [10ec:8136] (rev 02)
```

Realtek PCI network card not detected because there's no driver.
Appreciate if it can be added

```
Network controller [0280]: Qualcomm Atheros AR9285 Wireless Network Adapter (PCI-Express) [168c:002b] (rev 01)
```
Atheros AR9285 detected as wlan0 but cannot work as client mode.
AP mode is working fine.

Tested on Lenovo S10-3 netbook
```
DISTRIB_ID='OpenWrt'                                    DISTRIB_RELEASE='SNAPSHOT'                              DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'                                   DISTRIB_TAINTS='no-all override'                        DISTRIB_REVISION='R22.8.2'                              DISTRIB_DESCRIPTION='OpenWrt '                          NAME="OpenWrt"                                          VERSION="SNAPSHOT"
ID="openwrt"                                            ID_LIKE="lede openwrt"                                  PRETTY_NAME="OpenWrt SNAPSHOT"                          VERSION_ID="snapshot"                                   HOME_URL="https://openwrt.org/"                         BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"                BUILD_ID="r4818-114bd4f33"                              OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"                                   OPENWRT_TAINTS="no-all override"                        OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"  OPENWRT_DEVICE_PRODUCT="Generic"                        OPENWRT_DEVICE_REVISION="v0"                            OPENWRT_RELEASE="OpenWrt SNAPSHOT r4818-114bd4f33"      Linux OpenWrt 5.15.59 #0 SMP Wed Aug 10 09:40:11 2022 x86_64 GNU/Linux
```

haiibo, 2022-08-16:
Hello, the LEDE source code is not driven by R8101. You can try to use other source code compilation, or fork the source code and add the R8101 driver to compile

umarizulkifli, 2022-08-16:
Yes I'm trying to do that but I'm not sure if it's correct.
I'm still a beginner.
I found the source here http://mirror2.openwrt.org/sources/r8101-1.014.00.tar.bz2

haiibo, 2022-08-16:
Sorry, I won't know the OpenWrt source code, you can ask [ImmortalWrt](https://github.com/immortalwrt/immortalwrt) or [Lede](https://github.com/coolsnowwolf/lede)

umarizulkifli, 2022-08-18:
Can enable kmod-r8101
It's already in LEDE repo

haiibo, 2022-08-20:
I saw this driver, and I have added to config, just wait for the compile to be successful