haiibo/OpenWrt #24: REQUEST

ikhsanh, 2022-09-16:
HI. I'm using a tismart-ds866 and also a Huawei Me909s-821 Modem. My device doesn't support dual ethernet (lan+wan) so I can't connect to the internet. can you add some packages so that my huawei Me909s-821 modem can be detected on the interface? just add a few packages ( kmod-mii kmod-usb-net kmod-usb-wdm kmod-usb-net-qmi-wwan uqmi luci-proto-qmi
kmod-usb-net-cdc-ether kmod-usb-serial-option kmod-usb-serial kmod-usb-serial-wwan qmi-utils
kmod-usb-serial-qualcomm kmod-usb-acm kmod-usb-net-cdc-ncm kmod-usb-net-cdc-mbim umbim
modemmanager luci-proto-modemmanager usbutils ) I've tried to install it manually but failed.

haiibo, 2022-09-16:
你是什么设备，x86?

ikhsanh, 2022-09-16:
> 

Yes. X86

haiibo, 2022-09-16:
Sorry, I added network card driver to the RockChip platform last time. I forgot to add it to the X86 platform. I add now

ikhsanh, 2022-09-16:
> Sorry, I added network card driver to the RockChip platform last time. I forgot to add it to the X86 platform. I add now

Oke. Thanks

closed 2022-09-16 by ikhsanh