haiibo/OpenWrt #25: 请能问能添加一个R2S的 RTL8812BU的USB无线网卡驱动吗?

1335344215, 2022-09-16:
大佬:
请问能添加一个R2S的 RTL8812BU的USB无线网卡驱动吗?
具体芯片见图片
[image]

haiibo, 2022-09-16:
L大源码没有 rtl88x2bu 驱动，https://github.com/coolsnowwolf/lede/issues/9340

1335344215, 2022-09-16:
好吧,好像是内核版本的问题,5.4.162内核是支持的,后面找遍了全网都没找到支持最新内核又支持这款网卡的[image]

closed 2022-09-16 by 1335344215

haiibo, 2022-09-16:
你可以尝试拉取 immortalwrt  [openwrt-18.06-k5.4] (https://github.com/immortalwrt/immortalwrt/tree/openwrt-18.06-k5.4/package/kernel/rtl88x2bu) 或者 [openwrt-18.06](https://github.com/immortalwrt/immortalwrt/tree/openwrt-18.06/package/kernel/rtl88x2bu) 编译5.4最新内核和5.15试试看是否可用