haiibo/OpenWrt #26: 求助，辛苦老哥了

aoxueos, 2022-09-17:
老哥我模仿你的文件写法仿写了一个yml但是执行的时候报
/home/runner/work/_temp/386dc1a8-0fc0-4203-8c9b-9506b9b063b7.sh: /home/runner/work/OpenWrtBuild_Occupied/OpenWrtBuild_Occupied/diy-script.sh: /bin/bash^M: bad interpreter: No such file or directory
Error: Process completed with exit code 126.
救救孩子，地址https://github.com/aoxueos/OpenWrtBuild_Occupied
另外求解两个问题
* 如何添加sfe支持而不使用off
* 为啥你的Action没过但是能生成固件

haiibo, 2022-09-17:
源码10号过后大改了，编译占用空间费时长，Action 没过是插件多空间不够用或者需要编译的时间太长，Action 编译最长时间6个小时，到时间没编译好就会被取消

haiibo, 2022-09-17:
我因为超时很久没编译过了，我就本地编译上传的，准备精简一些插件，现在太过于臃肿了

haiibo, 2022-09-17:
[image]
上面的软件流量分载是off
下面的是sfe

haiibo, 2022-09-17:
[image]
diy-script.sh 文件格式不对，重新创建或者下载我的这个文件修改

aoxueos, 2022-09-17:
谢谢大佬的回复。我下载的就是您的R2S固件，并没有SFE。
[image]
然后我使用的diy-script.sh就是从您的项目下载的，但是仍然不起作用

haiibo, 2022-09-17:
我才发现 Rockchip 平台没添加，下面两条勾选了就有了
CONFIG_PACKAGE_kmod-shortcut-fe=y
CONFIG_PACKAGE_kmod-shortcut-fe-cm=y

aoxueos, 2022-09-17:
233

aoxueos, 2022-09-17:
话说可不可以用编译缓存吧一个编译过程拆成几部分来进行

haiibo, 2022-09-17:
现在功能还少啊，插件这么多了，再加只能本地编译了，本地编译 VMware 安装 Ubuntu 编译
我主页的本地提取有的，第五步不要执行 make menuconfig，把配置文件复制到 lede 目录内改名为 .config 执行make defconfig，后面用L大主页第四步编译就可以了

haiibo, 2022-09-17:
可以开始只编译IPK插件包，人后在生成固件

aoxueos, 2022-09-17:
还是报错，已经用您项目里的yml和diy了

haiibo, 2022-09-17:
你改的 yal 文件，diy-script.sh 你没动，是报这个文件找不到，这个文件删除了重新上传或者创建

aoxueos, 2022-09-17:
yal？新上传的yml文件是您项目里的一点没动

haiibo, 2022-09-17:
不对打错了，yml文件，其实你只要重新创建 diy-script.sh 文件就可以了，yml文件不用改

aoxueos, 2022-09-17:
yml和diy-script都没改啊，就改了configs里的rockship

haiibo, 2022-09-17:
你在网页上直接删除 diy-script.sh 重新上传，不要用 git，git 只识别内容

aoxueos, 2022-09-17:
好

aoxueos, 2022-09-17:
好了，成功进入编译环节，现在就研究研究怎么本地编译超大固件就好了。

aoxueos, 2022-09-17:
谢谢大佬

aoxueos, 2022-09-17:
git有毒

aoxueos, 2022-09-17:
话说老哥有空编译一般带sfe的r2s可好，谢谢老哥了

haiibo, 2022-09-17:
ok，添加代码了，看在线编译会不会超时

haiibo, 2022-09-18:
固件已上传，你可以测试一下是否有sfe

aoxueos, 2022-09-18:
> 固件已上传，你可以测试一下是否有sfe

sfe工作正常，但是经过测速开启sfe会导致网速控制和流量信息失效

closed 2022-09-18 by aoxueos