haiibo/OpenWrt #30: x96 max plus no wifi show up

karnadii, 2022-10-04:
no wifi for s905x3 x96 max plus for kernel 5.15 and 5.19

haiibo, 2022-10-04:
Do you have a wireless option in your network

karnadii, 2022-10-04:
> 

yes I have wireless option in network, but can not enable it.

haiibo, 2022-10-04:
You try to change the channel to see if it can be used normally

haiibo, 2022-10-04:
If it still cannot be used, it may be caused by the configuration change between the new production and the previous production. It is not recommended to use the built-in wifi.

karnadii, 2022-10-04:
[image]
[image]

can not change anything. I am not using the wifi as my main wifi. just for debugging purpose. If my modem has problem I can use wifi as WWAN temporarily like that.

haiibo, 2022-10-04:
Let me ask flippy,I use flippy's config file

haiibo, 2022-10-04:
you try https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_Aml_lede_10.04.0530/openwrt_s905x3_k5.15.71_2022.10.04.img.gz

haiibo, 2022-10-04:
See if wifi can be used

karnadii, 2022-10-04:
I have use that, and the wifi is working

haiibo, 2022-10-04:
That should be that the wireless driver is not added,Do you know your wireless network card model

karnadii, 2022-10-04:
rtl8822cs

haiibo, 2022-10-04:
Can you use wifi firmware before is lede source code?

karnadii, 2022-10-04:
ok let me try again to reflash the openwrt_aml_lede to make sure

karnadii, 2022-10-04:
no, sorry again. my x96 max plus suddenly won't boot from sdcard anymore. can't test again until I fix this

haiibo, 2022-10-04:
You need to compile the official firmware, I use lede, lede does not have 8822 driver

karnadii, 2022-10-04:
```
0-03
───────────────────────────────────────────────────────────────────────
 Device Model: X96 MAX PLUS Q2, X96 Air Q1000
 Architecture: AArch64 : Cortex-A55 x 4
 Load Average: 0.04 0.19 0.11   Uptime: 5m 11s
 Ambient Temp: 61.8  °C         CPU Freq: 1800 Mhz
 Memory Usage: 5% of 3337M      Swap Usage: 0% of 0M   
 Boot Storage: 39% of 254.7M    ROOTFS: 87% of 960.0M
───────────────────────────────────────────────────────────────────────
root@OpenWrt:~# lsusb
Bus 001 Device 008: ID 05c6:6000 Qualcomm, Incorporated Qualcomm CDMA Technologies MSM
Bus 002 Device 001: ID 1d6b:0003 Linux 5.15.71-flippy-77+o xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0003 Linux 5.15.71-flippy-77+o vhci_hcd USB/IP Virtual Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.71-flippy-77+o xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0002 Linux 5.15.71-flippy-77+o vhci_hcd USB/IP Virtual Host Controller
root@OpenWrt:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_TARGET='armvirt/64'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_TAINTS='no-all'
DISTRIB_SOURCECODE='lede'
DISTRIB_REVISION='R2022.09.29'
DISTRIB_DESCRIPTION='OpenWrt '
root@OpenWrt:~# 
```
this one has a working wifi. this is lede right?

haiibo, 2022-10-05:
You try the newly compiled firmware, I added some wireless drivers

karnadii, 2022-10-05:
> You try the newly compiled firmware, I added some wireless drivers

still not working. it is not crucial, so since this is not supported by lede I will close it, thanks for your time.

closed 2022-10-05 by karnadii