haiibo/OpenWrt #35: cm311-1a可以编译专用的科学固件吗

97668589, 2022-10-25:
cm311-1a可以编译专用的科学固件吗

haiibo, 2022-10-25:
[mini](https://github.com/haiibo/OpenWrt/releases/download/ARMv8_MINI/openwrt_s905x2_x96max_R22.10.10_k5.15.74-flippy-78+o.img.gz) 版本就是主打科学上网的

haiibo, 2022-10-26:
下载 [openwrt_s905x2_x96max_R22.10.10_k5.15.74-flippy-78+o.img.gz](https://github.com/haiibo/OpenWrt/releases/download/ARMv8_MINI/openwrt_s905x2_x96max_R22.10.10_k5.15.74-flippy-78+o.img.gz) 按照 [cm311-1a刷f大openwrt并写入emmc](https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8259393&extra=page%3D1%26filter%3Dlastpost%26orderby%3Dlastpost) 刷入

97668589, 2022-10-26:
U盘启动行不行
 

随心网
***@***.***

 

 
&mdash;

97668589, 2022-10-26:
测试了，这个固件刷入U盘后无法启动！

haiibo, 2022-10-26:
是按照上面的方法刷入的吗

haiibo, 2022-10-26:
或者 fork 项目，修改 [ARMv8-Mini-OpenWrt.yml](https://github.com/haiibo/OpenWrt/blob/main/.github/workflows/ARMv8-Mini-OpenWrt.yml) 文件内 
    - name: Package OpenWrt Firmware
      if: steps.download.outputs.status == 'success'
      uses: unifreq/openwrt_packit@master
      env:
        OPENWRT_ARMVIRT: openwrt-armvirt/*rootfs.tar.gz
        PACKAGE_SOC: all
        WHOAMI: Haiibo
        KERNEL_REPO_URL: https://github.com/ophub/kernel/tree/main/pub/stable
        KERNEL_VERSION_NAME: 5.15.13_6.0.2
        KERNEL_AUTO_LATEST: true
        DISTRIB_REVISION: R${{ env.DATE1 }} by Haiibo
改为
    - name: Package OpenWrt Firmware
    - if: steps.download.outputs.status == 'success'
      uses: ophub/amlogic-s9xxx-openwrt@main
      with:
        openwrt_path: openwrt-armvirt/*rootfs.tar.gz
        openwrt_soc: s905l3a
        openwrt_kernel: 5.10.125_5.15.50
然后编译即可

97668589, 2022-10-26:
我试试

haiibo, 2022-10-26:
你前面刷入U盘有修改dtb吗，刷入U盘后修改 /BOOT/uEnv.txt 文件，把 meson-g12a-x96-max.dtb 改成 meson-g12a-s905l3a-cm311.dtb

97668589, 2022-10-26:
/BOOT/uEnv.txt这个文件在哪里
 

随心
***@***.***

97668589, 2022-10-26:
https://github.com/ophub/amlogic-s9xxx-openwrt
老大，你能不能直接在这个项目添加科学啊，这个固件可以正常启动，就是没有科学！主要是我不会编译固件！

haiibo, 2022-10-26:
刷入U盘后，重新把U盘插入电脑会有BOOT文件夹里面

97668589, 2022-10-26:
我去，还真有额，这个和别的固件不一样，别的固件刷入U盘打不开了！

haiibo, 2022-10-26:
你改一下dtb，修改完成应该能成功启动

97668589, 2022-10-26:
好的，我试试

haiibo, 2022-10-26:
> https://github.com/ophub/amlogic-s9xxx-openwrt 老大，你能不能直接在这个项目添加科学啊，这个固件可以正常启动，就是没有科学！主要是我不会编译固件！

在 [diy-part2.sh](https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/router-config/lede-master/diy-part2.sh) 文件内添加

# 科学上网插件
git clone --depth 1 https://github.com/jerrykuku/luci-app-vssr package/luci-app-vssr
git clone --depth 1 https://github.com/jerrykuku/lua-maxminddb package/lua-maxminddb
svn co https://github.com/kiddin9/openwrt-bypass/trunk/luci-app-bypass package/luci-app-bypass
svn co https://github.com/vernesong/OpenClash/trunk/luci-app-openclash package/luci-app-openclash
svn co https://github.com/xiaorouji/openwrt-passwall/branches/luci/luci-app-passwall package/luci-app-passwall
svn co https://github.com/xiaorouji/openwrt-passwall2/trunk/luci-app-passwall2 package/luci-app-passwall2
svn co https://github.com/fw876/helloworld/trunk/luci-app-ssr-plus package/luci-app-ssr-plus

# 科学上网插件依赖
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/brook package/brook
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/chinadns-ng package/chinadns-ng
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/dns2socks package/dns2socks
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/dns2tcp package/dns2tcp
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/hysteria package/hysteria
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/ipt2socks package/ipt2socks
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/microsocks package/microsocks
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/naiveproxy package/naiveproxy
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/pdnsd-alt package/pdnsd-alt
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/sagernet-core package/sagernet-core
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/ssocks package/ssocks
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/tcping package/tcping
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/trojan-go package/trojan-go
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/trojan-plus package/trojan-plus
svn co https://github.com/xiaorouji/openwrt-passwall/trunk/v2ray-geodata package/v2ray-geodata
svn co https://github.com/fw876/helloworld/trunk/simple-obfs package/simple-obfs
svn co https://github.com/fw876/helloworld/trunk/v2ray-core package/v2ray-core
svn co https://github.com/fw876/helloworld/trunk/v2ray-plugin package/v2ray-plugin
svn co https://github.com/fw876/helloworld/trunk/shadowsocks-rust package/shadowsocks-rust
svn co https://github.com/fw876/helloworld/trunk/shadowsocksr-libev package/shadowsocksr-libev
svn co https://github.com/fw876/helloworld/trunk/xray-core package/xray-core
svn co https://github.com/fw876/helloworld/trunk/xray-plugin package/xray-plugin
svn co https://github.com/fw876/helloworld/trunk/lua-neturl package/lua-neturl
svn co https://github.com/fw876/helloworld/trunk/trojan package/trojan

97668589, 2022-10-26:
大佬，终于可以了，非常-----感谢!--- 感谢! ---感谢!

closed 2022-10-26 by haiibo