haiibo/OpenWrt #37: 关于精简版的问题

lzqd, 2022-10-26:
关于精简版的问题，貌似没有装ntfs的相关，比如ntfs3，希望能添加

haiibo, 2022-10-26:
如需挂载U盘或硬盘建议使用 PLUS 版固件

lzqd, 2022-10-26:
> 

PLUS 版固件没有s905x3的固件

lzqd, 2022-10-26:
> 如需挂载U盘或硬盘建议使用 PLUS 版固件

PLUS 版固件经常会编译失败。

haiibo, 2022-10-26:
> > 如需挂载U盘或硬盘建议使用 PLUS 版固件
> 
> PLUS 版固件经常会编译失败。

你是本地编译还是云编译，有日志吗

haiibo, 2022-10-26:
我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时
Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673

lzqd, 2022-10-26:
> 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673

PLUS 版固件编译失败是超时，具体什么原因我也不清楚
精简版使用你这下的缺少ntfs相关也无法安装。

haiibo, 2022-10-26:
> > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> 
> PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。

编译超时就是插件选太多，分配的CPU性能不足
`Check Server Performance`步骤就是检查CPU
[image]

haiibo, 2022-10-26:
> > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> 
> PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。

你是用mini版opkg安装插件提示缺少ntfs吗

lzqd, 2022-10-26:
> > > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> > 
> > 
> > PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。
> 
> 你是用mini版opkg安装插件提示缺少ntfs吗

ntfs3-mount和kmod-fs-ntfs3两个无法安装，其它东西不缺少，没有这两个，磁盘无法识别

lzqd, 2022-10-26:
> > > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> > 
> > 
> > PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。
> 
> 你是用mini版opkg安装插件提示缺少ntfs吗

提示版本和内核不匹配什么的

haiibo, 2022-10-26:
> > > > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> > > 
> > > 
> > > PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。
> > 
> > 
> > 你是用mini版opkg安装插件提示缺少ntfs吗
> 
> 提示版本和内核不匹配什么的

你使用的什么源

lzqd, 2022-10-26:
> 

你这下的精简版

lzqd, 2022-10-26:
> > > > > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> > > > 
> > > > 
> > > > PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。
> > > 
> > > 
> > > 你是用mini版opkg安装插件提示缺少ntfs吗
> > 
> > 
> > 提示版本和内核不匹配什么的
> 
> 你使用的什么源，有的有区别会不匹配

你这下的精简版，什么都没改

lzqd, 2022-10-26:
> 

什么都没改

lzqd, 2022-10-26:
> > > > > > > 我的经常编译失败是因为插件选的太多，云编译必须分配到8370C，8272CL这两种CPU才会编译完成，如果是8171M，E5-2673这两个CPU性能太差会导致编译超时 Github Actions已知CPU型号（降序）：8370C，8272CL，8171M，E5-2673
> > > > > > 
> > > > > > 
> > > > > > PLUS 版固件编译失败是超时，具体什么原因我也不清楚 精简版使用你这下的缺少ntfs相关也无法安装。
> > > > > 
> > > > > 
> > > > > 你是用mini版opkg安装插件提示缺少ntfs吗
> > > > 
> > > > 
> > > > 提示版本和内核不匹配什么的
> > > 
> > > 
> > > 你使用的什么源，有的有区别会不匹配
> > 
> > 
> > 你这下的精简版，什么都没改
> 
> 我是说你opkg的源，有没有修改

没有

haiibo, 2022-10-26:
> 

我添加了，等待明天上午编译的固件

lzqd, 2022-10-26:
> 加了，等待明天上午编译的固件

谢了

closed 2022-10-26 by haiibo