haiibo/OpenWrt #72: 关于S905L3A机型的请教

GD2021, 2023-03-26:
您好，我的机顶盒为创维E900V22D；cup型号是S905L3A我现在已经刷了armbian，版本是Armbian_23.05.0_amlogic_s905l3a_bullseye_5.15.104_server_2023.03.24.img.gz。
[image]

问题一：可以用您的哪个OP固件呢？

问题二：我想在KVM里装openwrt，请问下载您的哪个qcow2包？

ysuolmai, 2023-03-27:
随便下一个S905X2或者X3的固件，然后改uEnv.txt里的dtb。
或者，直接把你的armbian里/boot/uEnv.txt 复制到openwrt里的/boot目录，覆盖，就完事了。

GD2021, 2023-03-27:
> 随便下一个S905X2或者X3的固件，然后改uEnv.txt里的dtb。
> 
> 或者，直接把你的armbian里/boot/uEnv.txt 复制到openwrt里的/boot目录，覆盖，就完事了。

那KVM包呢

ysuolmai, 2023-03-27:
> > 随便下一个S905X2或者X3的固件，然后改uEnv.txt里的dtb。
> > 或者，直接把你的armbian里/boot/uEnv.txt 复制到openwrt里的/boot目录，覆盖，就完事了。
> 
> 那KVM包呢

一共就两个qcow2文件，试试就知道了

closed 2023-04-18 by haiibo