haiibo/OpenWrt #86: 遭遇断流，系统日志：bpfilter: read fail -512

wordlesswind, 2023-05-27:
设备是电犀牛 R68S，固件是 [R2023-05-27 02:26:03 for ARMv8_PLUS](https://github.com/haiibo/OpenWrt/releases/tag/ARMv8_PLUS)

系统日志看不出是什么问题，前面根本没有相关日志，重启 LAN 接口即可恢复网络。

只开了 Turbo ACC 网络加速、UPnP 和 MosDNS，现在把 MosDNS 关了，然后启用 的`启用额外的日志`看看有没有额外日志。

系统日志：

```
...
Sat May 27 16:52:28 2023 kern.err kernel: [24911.562056] bpfilter: read fail -512
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:47 2023 daemon.err miniupnpd[28071]: sendto(udp_notify=6, 192.168.2.1): Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=462, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=478, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=460, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=446, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=466, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=407, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=470, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto(sock=6, len=398, dest=239.255.255.250:1900): sendto: Operation not permitted
Sat May 27 16:52:48 2023 daemon.err miniupnpd[28071]: try_sendto failed to send 11 packets
...
```

网上能找到的信息只有一个帖子：https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4053183&ordertype=1&page=7

haiibo, 2023-05-27:
我没遇到这种情况，是使用的最新固件吗，你尝试一下[ARMv8_MINI](https://github.com/haiibo/OpenWrt/releases/download/ARMv8_MINI/openwrt_rk3568_h68k_R23.4.1_k6.1.29-flippy-83+.img.gz)版本和[Rockchip](https://github.com/haiibo/OpenWrt/releases/download/Rockchip/openwrt-rockchip-armv8-hinlink_opc-h68k-squashfs-sysupgrade.img.gz)直接编译版本试试

wordlesswind, 2023-05-27:
> 之前有没有出现过这个问题，还是更新固件后才出现的

刚买🤣，看到[电犀牛R66s 固件推荐 - 大侠阿木博客](https://www.daxiaamu.com/7476/)就刷了。

haiibo, 2023-05-27:
目前我没有遇到断流情况，你先试试[ARMv8_MINI](https://github.com/haiibo/OpenWrt/releases/download/ARMv8_MINI/openwrt_rk3568_h68k_R23.4.1_k6.1.29-flippy-83+.img.gz)或者LEDE直接编译版本[Rockchip](https://github.com/haiibo/OpenWrt/releases/download/Rockchip/openwrt-rockchip-armv8-fastrhino_r68s-squashfs-sysupgrade.img.gz)看看

wordlesswind, 2023-05-27:
现在刷了 [LEDE 直接编译](https://github.com/haiibo/OpenWrt/releases/tag/Rockchip)版本，等等看效果。

但是有个问题，似乎 LEDE 直接编译版本导致 Recovery 按钮失效了，这意味着不能线刷了，那后续我能够手动安装[晶晨宝盒](https://github.com/ophub/luci-app-amlogic/tree/main)然后刷 [ARMv8_MINI](https://github.com/haiibo/OpenWrt/releases/tag/ARMv8_MINI) 或者其他版本吗？

haiibo, 2023-05-27:
我没有这样尝试过，不清楚行不行。

haiibo, 2023-05-27:
你使用多久断流的，是r68s拨号吗

wordlesswind, 2023-05-27:
> 你使用多久断流的，是r68s拨号吗

大概是七、八个小时吧，对的。

wordlesswind, 2023-05-27:
> 我没有这样尝试过，不清楚行不行。

看样子不行，这个直接编译版本似乎没有适配。

```
...
Sat May 27 19:37:14 2023 daemon.err uhttpd[2271]: lsblk: /dev/mmcblk02: not a block device
...
Sun May 28 00:55:59 2023 daemon.err uhttpd[2271]: ERROR: not a btrfs filesystem: /
Sun May 28 00:55:59 2023 daemon.err uhttpd[2271]: ERROR: can't access '/'
...
```

直接编译版本不像 flippy 的版本使用 btrfs，也没有进行分区调整，不是的是哪个插件试图访问 `/dev/mmcblk02`。

[image]
[image]

它只有三个分区，挂载分区用的 ext4，还是 MBR 的。

```
...
Sat May 27 18:48:26 2023 daemon.err nginx[9918]: 2023/05/27 18:48:24 [emerg] 9918#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Sat May 27 18:48:26 2023 daemon.err nginx[9918]: 2023/05/27 18:48:24 [emerg] 9918#0: bind() to [::]:443 failed (98: Address in use)
Sat May 27 18:48:26 2023 daemon.err nginx[9918]: 2023/05/27 18:48:24 [emerg] 9918#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Sat May 27 18:48:26 2023 daemon.err nginx[9918]: 2023/05/27 18:48:24 [emerg] 9918#0: bind() to [::]:80 failed (98: Address in use)
Sat May 27 18:48:27 2023 daemon.err nginx[9918]: 2023/05/27 18:48:24 [emerg] 9918#0: still could not bind()
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:443 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:80 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:443 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Sat May 27 18:48:32 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:80 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:443 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:80 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:443 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Sat May 27 18:48:33 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:80 failed (98: Address in use)
Sat May 27 18:48:34 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:443 failed (98: Address in use)
Sat May 27 18:48:34 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:443 failed (98: Address in use)
Sat May 27 18:48:34 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Sat May 27 18:48:34 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: bind() to [::]:80 failed (98: Address in use)
Sat May 27 18:48:34 2023 daemon.err nginx[10754]: 2023/05/27 18:48:32 [emerg] 10754#0: still could not bind()
...
Sat May 27 18:48:46 2023 daemon.notice procd: /etc/rc.d/S99serverchan: serverchan is starting now ...
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib confmisc.c:855:(parse_card) cannot find card '0'
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_card_inum returned error: No such file or directory
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib confmisc.c:422:(snd_func_concat) error evaluating strings
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib confmisc.c:1334:(snd_func_refer) error evaluating name
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib conf.c:5701:(snd_config_expand) Evaluate error: No such file or directory
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib control.c:1528:(snd_ctl_open_noupdate) Invalid CTL default
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: amixer: Mixer attach default error: No such file or directory
[... cut ...]
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib conf.c:5701:(snd_config_expand) Evaluate error: No such file or directory
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: ALSA lib control.c:1528:(snd_ctl_open_noupdate) Invalid CTL default
Sat May 27 18:48:47 2023 daemon.notice procd: /etc/rc.d/S99shairport-sync: amixer: Mixer attach default error: No such file or directory
...
```

其他的报错

haiibo, 2023-05-28:
你用大雕的固件【系统 → 备份/升级】更新一下[R22.11.13 Release : Let's Rock !](https://github.com/coolsnowwolf/lede/releases/tag/20220716)试试能不能用mask按键，这是老版内核，记得把保留配置勾取消掉

wordlesswind, 2023-05-28:
> 你用大雕的固件【系统 → 备份/升级】更新一下[R22.11.13 Release : Let's Rock !](https://github.com/coolsnowwolf/lede/releases/tag/20220716)试试能不能用mask按键，这是老版内核，记得把保留配置勾取消掉

我先看看还会不会断流吧，也许可能是 flippy 大佬的新内核有问题。

haiibo, 2023-05-28:
我昨晚安装PLUS版本最新固件到现在12个小时没有出现断流问题
[image]

wordlesswind, 2023-05-28:
直接编译版本也没有出现 `bpfilter: read fail` 情况，就先这样吧，懒得折腾了。

closed 2023-05-28 by wordlesswind

wordlesswind, 2023-05-29:
> 你用大雕的固件【系统 → 备份/升级】更新一下[R22.11.13 Release : Let's Rock !](https://github.com/coolsnowwolf/lede/releases/tag/20220716)试试能不能用mask按键，这是老版内核，记得把保留配置勾取消掉

尝试了下，不行。

```
Mon May 29 09:31:55 2023 daemon.err uhttpd[2271]: Mon May 29 09:31:55 CST 2023 upgrade: Device fastrhino,r68s not supported by this image
Mon May 29 09:31:55 2023 user.info upgrade: Device fastrhino,r68s not supported by this image
Mon May 29 09:31:55 2023 daemon.err uhttpd[2271]: Mon May 29 09:31:55 CST 2023 upgrade: Supported devices: rockchip,rk3568-lunzn-r68s
Mon May 29 09:31:55 2023 user.info upgrade: Supported devices: rockchip,rk3568-lunzn-r68s
Mon May 29 09:31:55 2023 daemon.err uhttpd[2271]: Mon May 29 09:31:55 CST 2023 upgrade: Reading partition table from bootdisk...
Mon May 29 09:31:55 2023 user.info upgrade: Reading partition table from bootdisk...
Mon May 29 09:31:55 2023 daemon.err uhttpd[2271]: cat: write error: Broken pipe
Mon May 29 09:31:55 2023 daemon.err uhttpd[2271]: Mon May 29 09:31:55 CST 2023 upgrade: Reading partition table from image...
Mon May 29 09:31:55 2023 user.info upgrade: Reading partition table from image...
Mon May 29 09:31:55 2023 daemon.err uhttpd[2271]: Mon May 29 09:31:55 CST 2023 upgrade: Invalid partition table on /tmp/image.bs
Mon May 29 09:31:55 2023 user.info upgrade: Invalid partition table on /tmp/image.bs
...
Mon May 29 09:33:38 2023 user.info upgrade: Device fastrhino,r68s not supported by this image
Mon May 29 09:33:38 2023 user.info upgrade: Supported devices: rockchip,rk3568-lunzn-r68s
Mon May 29 09:33:38 2023 user.info upgrade: Reading partition table from bootdisk...
Mon May 29 09:33:38 2023 user.info upgrade: Reading partition table from image...
Mon May 29 09:33:38 2023 user.info upgrade: Invalid partition table on /tmp/image.bs
Mon May 29 09:33:39 2023 user.info upgrade: Commencing upgrade. Closing all shell sessions.
Mon May 29 09:33:39 2023 user.info upgrade: Device fastrhino,r68s not supported by this image
Mon May 29 09:33:39 2023 user.info upgrade: Supported devices: rockchip,rk3568-lunzn-r68s
Mon May 29 09:33:39 2023 user.info upgrade: Reading partition table from bootdisk...
Mon May 29 09:33:39 2023 user.info upgrade: Reading partition table from image...
Mon May 29 09:33:39 2023 user.info upgrade: Invalid partition table on /tmp/image.bs
```

wordlesswind, 2023-05-29:
……没事了，官方给的方法是：

> 如果刷入固件后无法进入 Web 界面，可以通过 SSH 连接路由器后，输入命令 `firstboot`
> 
> 刷入`5.1x.y` 以上内核固件后，如果想进入刷机模式，可以通过 SSH 连接后，输入命令 `dd if=/dev/zero of=/dev/mmcblk0 bs=8M count=1`，然后重启，可以进入 MASRKROM 刷机模式.。

直接格式化可还行吧……