haiibo/OpenWrt #88: 关闭LAN的桥接接口之后，信息无法正常显示

Cod1ng, 2023-06-06:
使用环境：qnap nas TS-464C 虚拟机安装
版本：OpenWrt R23.06.04 by Haiibo / LuCI Master (git-23.141.16773-28dd4b3)
遇到的问题：
在网络-接口-LAN-物理设置，取消掉“桥接接口”的选项之后，网络-接口会一直显示“正在收集数据..."
[image]
[image]
只要勾选上，就能正常显示
[image]
试过别的好几个openwrt，都没有这个问题，但别的固件除了没这个问题以外，都没有你的好用，求修复，需要什么别的日志也可以告知

haiibo, 2023-06-07:
重启

Cod1ng, 2023-06-07:
重启之后的确正常了，感谢

closed 2023-06-07 by Cod1ng