haiibo/OpenWrt #94: 求新增机型 panther-x2

xiaomeng9597, 2023-06-22:
设备名称：PantherX2 Hotspot（黑豹X2）

官网：https://shop.panther.global/

用户指南：https://g.itemz.cn/sgwBfS

配置：
处理器: 瑞芯微，RK3566，四核 Cortex-A55 高达 1.8GHz
内存：DDR 4GB
储存：eMMC 32GB+ TF卡64GB

别人在ophub大佬的云编译提交的反馈
https://github.com/ophub/amlogic-s9xxx-armbian/issues/1319

haiibo, 2023-06-29:
暂不考虑添加，如果确实需要可以使用 fork 项目后修改`ARMv8-Package.yml`文件

将`$GITHUB_REPOSITORY`改为`haiibo/OpenWrt`，或者自行编译固件后打包

然后修改
```
    - name: Package OpenWrt Firmware
      if: steps.download.outputs.status == 'success'
      uses: ophub/flippy-openwrt-actions@main
      env:
        OPENWRT_ARMVIRT: openwrt/bin/targets/*/*/*rootfs.tar.gz
        PACKAGE_SOC: all
        WHOAMI: Haiibo
        KERNEL_VERSION_NAME: 5.15.13_6.1.5
        KERNEL_AUTO_LATEST: true
        DISTRIB_REVISION: R${{ env.DATE1 }} by Haiibo
```
改为
```
    - name: Package OpenWrt Firmware
      if: steps.download.outputs.status == 'success'
      uses: ophub/amlogic-s9xxx-openwrt@main
      with:
        openwrt_path: openwrt/bin/targets/*/*/*rootfs.tar.gz
        openwrt_board: panther-x2
        openwrt_kernel: 5.15.13_6.1.5
        auto_kernel: true
        kernel_repo: ophub/kernel
```
然后运行该 workflows 自行打包

xiaomeng9597, 2023-06-29:
谢谢大佬回复

closed 2023-07-01 by haiibo