haiibo/OpenWrt #95: MINI版本是否有Docker镜像？

Java2Boy, 2023-06-23:
环境 ：N1 Armbian 23.08.0-trunk Bullseye with Linux 6.1.34-ophub  + Docker

是否有适用 N1 MINI版镜像，求发布或镜像链接，非常感谢

iJayscn, 2023-06-25:
同问，N1只要需要mini版 

├ VSSR 
├ SSR-Plus 
├ Passwall 
├ OpenClash 
├ DDNS 
├ ipv6
├ snmpd
├ iperf3
├ Wireguard
└ Zerotier

Java2Boy, 2023-06-25:
> 
可以试试summary/openwrt-aarch64:latest，但这版固件安装speederv2,udp2raw无法安装，不知道怎么回事

haiibo, 2023-06-29:
无法安装是因为缺少依赖，找全依赖，用ipkg命令安装

haiibo, 2023-06-29:
MINI 版固件上传到 Docker 了，summary/openwrt-aarch64:mini

haiibo, 2023-06-29:
因为这个固件是原本给本地使用的固件，我是直接用MINI版底包打包上传的，有很多东西在docker环境是多余的，所以比较大。原本那个docker固件是专门针对docker环境编译的，所以比较小

Java2Boy, 2023-07-02:
> 因为这个固件是原本给本地使用的固件，我是直接用MINI版底包打包上传的，有很多东西在docker环境是多余的，所以比较大。原本那个docker固件是专门针对docker环境编译的，所以比较小

 应用过滤（OpenAppFilter）是否能支持？ 手动安装IPK失败，是否需要增加内核级支持？

levin9590, 2023-08-04:
fork后修改哪个文件可以只打包生成需要的机型固件，剔除不需要的机型，节省时间和平台资源。