haiibo/OpenWrt #104: 网速下降

hometest1234, 2023-07-07:
刷了7月5号的x86版本，原本千M宽带，现在下载速度只有59M, 上行正常，这是什么原因，急！启用 SYN-flood 防御 已开启， 启用FullCone-NAT 为高性能模式 lan IP 动态伪装 也勾上了。 连接方式，路由器网线直连这个测试电脑，没有他外接设备。

ektest2023, 2023-07-07:
我也遇到类似的问题，fork了这个项目，只改动原登陆IP进行编译的，也是降速了。

ektest2023, 2023-07-31:
我还是懂怎么升级docker到最新版，我替换了docker dockerd 然后拉取别人库里的docker但是编译完成后，docker 菜单消失了，但编译找到安装软件还是20.0.17的版，求助一下大佬，能否你升给你的库，我重新fork一下，谢谢了

Message ID: ***@***.***>

haiibo, 2023-07-31:
在 `diy-script.sh` 文件内添加以下内容即可

```
# 更新 docker 版本
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=24.0.5/g' feeds/packages/utils/docker/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=fa32b5f3c2f85fba9ef6e1b5099a4b608fa20af45ba71b3da2194e8728037eec/g' feeds/packages/utils/docker/Makefile
sed -i 's/PKG_GIT_SHORT_COMMIT:=.*/PKG_GIT_SHORT_COMMIT:=ced0996/g' feeds/packages/utils/docker/Makefile

# 更新 dockerd 版本
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=24.0.5/g' feeds/packages/utils/dockerd/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=837d7d667fb64508bf6e53cb5915b4b5ef356599294ffdd5ca8678168230cb38/g' feeds/packages/utils/dockerd/Makefile
sed -i 's/PKG_GIT_SHORT_COMMIT:=.*/PKG_GIT_SHORT_COMMIT:=a61e2b4/g' feeds/packages/utils/dockerd/Makefile
```

ektest2023, 2023-08-01:
谢谢大佬，我试一下。

ektest2023, 2023-08-02:
大佬按上面的参数加进去后，一直编译报错，这是错误代码：make[2] -C feeds/packages/utils/docker compile
    ERROR: package/feeds/packages/docker failed to build.

haiibo, 2023-08-02:
你直接用老哥改好的吧
```
rm -rf feeds/packages/utils/docker
rm -rf feeds/packages/utils/dockerd
svn export https://github.com/breakings/OpenWrt/trunk/general/docker feeds/packages/utils/docker
svn export https://github.com/breakings/OpenWrt/trunk/general/dockerd feeds/packages/utils/dockerd
```

ektest2023, 2023-08-04:
用了老哥那个docker代码，编译二天了一直报错，请大佬出下手，你更新一下给我吧，谢谢，搞不定了。

ektest2023, 2023-08-12:
大佬有更新吗docker