haiibo/OpenWrt #108: 插件更新

ektest2023, 2023-07-17:
自编译，docker版本能否更新到最新版，怎么做才行？现在是20.10.17，docker最新版应该是：24.0.2-1。直接升级报错如下：Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-crypto-acompress
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-lzo
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-zlib-inflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-zlib-deflate
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-raid6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-xor
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-lib-zstd
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-fs-btrfs
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nft-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nft-compat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-compat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ipt-extra
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-ipt6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ip6tables
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-nat6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ipt-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ipt-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ipt-nat6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-br-netfilter
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-ipt-physdev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-physdev found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-ipvs
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-veth
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nft-fib
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-fib found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nf-flow
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nft-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0) for kmod-nft-offload
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-offload found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for dockerd:
 * 	kernel (= 5.15.120-1-457c33b95a1a9e5aafe27d4471ec94d0)
 * opkg_install_cmd: Cannot install package dockerd.

leyuanwl, 2023-07-17:
蓝牙模块安装也是 这个错 还有硬盘支持插件也是 ，应该是kernel版本的问题

ektest2023, 2023-07-17:
可以修复吗？另外固件网速度好像变慢了，J4125 主机。

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

leyuanwl, 2023-07-17:
参考这个 ： https://www.rstk.cn/news/1403933.html?action=onClick

closed 2023-07-31 by haiibo