haiibo/OpenWrt #112: Docker版本

hometest1234, 2023-07-29:
Docker版本落后几个版本了，更新也不行，编泽改其它版本如何做？

haiibo, 2023-07-29:
要使用最新的docker版本需要编译时自行更新docker文件夹内Makefile文件，把版本和HASH改为最新即可

closed 2023-07-31 by haiibo

haiibo, 2023-07-31:
在 `diy-script.sh` 文件内添加以下内容即可

```
# 更新 docker 版本
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=24.0.5/g' feeds/packages/utils/docker/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=fa32b5f3c2f85fba9ef6e1b5099a4b608fa20af45ba71b3da2194e8728037eec/g' feeds/packages/utils/docker/Makefile
sed -i 's/PKG_GIT_SHORT_COMMIT:=.*/PKG_GIT_SHORT_COMMIT:=ced0996/g' feeds/packages/utils/docker/Makefile

# 更新 dockerd 版本
sed -i 's/PKG_VERSION:=.*/PKG_VERSION:=24.0.5/g' feeds/packages/utils/dockerd/Makefile
sed -i 's/PKG_HASH:=.*/PKG_HASH:=837d7d667fb64508bf6e53cb5915b4b5ef356599294ffdd5ca8678168230cb38/g' feeds/packages/utils/dockerd/Makefile
sed -i 's/PKG_GIT_SHORT_COMMIT:=.*/PKG_GIT_SHORT_COMMIT:=a61e2b4/g' feeds/packages/utils/dockerd/Makefile
```